FUjr/QModem #6: 我这个编译加了Qmodem 然后主页显示有问题

JDKCHINA, 2024-11-30:
[image]
我然后去卸载之后就显示正常了，麻烦大佬修复一下，不过大佬的模组插件就是好用 麻烦大佬加下RM502Q-GL RM520F-BL RM510Q-GL  谢谢

FUjr, 2024-12-01:
这个是jsluci还是lua luci？lua luci的话可能是css翻车了

FUjr, 2024-12-01:
jsluci编译时不要选择“包含lua首页信息展示”
jsluci已经默认带了首页信息，lua的为了兼容lean的lua luci，添加了一些css 因此会破坏jsluci的样式

yaoyue12548, 2024-12-04:
[image]
大佬即使不选那个lua首页，模块信息显示这样，不过首页信息显示正常了。

FUjr, 2024-12-04:
> 
> 
> [image]
> 
> 大佬即使不选那个lua首页，模块信息显示这样，不过首页信息显示正常了。
> 
> 
这个是为了兼容luci18，这个可以接受吗，不能接受可以暂时在luci/luci-app-qmodem/luasrc/view/modem_overview.htm里删除和progress_bar相关的css，短期暂时没空修复

FUjr, 2024-12-04:
另外在jsluci的bootstrap主题似乎不存在这个问题

yaoyue12548, 2024-12-04:
好的，大佬在bootstrap没问题。感谢大佬

commit 2907ba88 (2024-12-19): luci: refactor progress bar class names and add compatibility check to fix #6
closed 2024-12-19 by FUjr