FUjr/QModem #7: 使用已经使用apk管理器Snapshot版本的openwrt编译tom_modem一定会报错，即使不勾选tom_modem和quectel_CM_5G_M

qwer155168, 2024-12-09:
time: package/feeds/luci/luci-theme-argon/compile#0.11#0.00#0.11
make[3]: Entering directory '/home/qwer/immortalwrt-ipq/feeds/modem/application/tom_modem'
mkdir -p /home/qwer/immortalwrt-ipq/staging_dir/target-aarch64_cortex-a53_musl/root-qualcommax/stamp
SHELL= flock /home/qwer/immortalwrt-ipq/tmp/.root-copy.flock -c 'cp -fpR /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/.pkgdir/tom_modem/. /home/qwer/immortalwrt-ipq/staging_dir/target-aarch64_cortex-a53_musl/root-qualcommax/'
touch /home/qwer/immortalwrt-ipq/staging_dir/target-aarch64_cortex-a53_musl/root-qualcommax/stamp/.tom_modem_installed
rm -rf /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem
mkdir -p /home/qwer/immortalwrt-ipq/bin/targets/qualcommax/ipq807x/packages /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem /home/qwer/immortalwrt-ipq/staging_dir/target-aarch64_cortex-a53_musl/pkginfo
install -d -m0755 /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/usr/bin
install -m0755 /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/tom_modem /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/usr/bin/
find /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/qwer/immortalwrt-ipq/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/qwer/immortalwrt-ipq/scripts/strip-kmod.sh" PATCHELF="/home/qwer/immortalwrt-ipq/staging_dir/host/bin/patchelf" /home/qwer/immortalwrt-ipq/scripts/rstrip.sh /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem
rstrip.sh: /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/usr/bin/tom_modem: executable
install -d -m0755 /home/qwer/immortalwrt-ipq/bin/packages/aarch64_cortex-a53/modem/tmp
mkdir -p /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/
mkdir -p /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/
(cd /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem; )
( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo 'export root="${IPKG_INSTROOT}"'; echo 'export pkgname="tom_modem"'; echo "add_group_and_user"; [ ! -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/postinst-pkg ] || cat "/home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/postinst-pkg"; echo "default_postinst"; ) > /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/post-install;
( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo 'export root="${IPKG_INSTROOT}"'; echo 'export pkgname="tom_modem"'; [ ! -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/prerm-pkg ] || cat "/home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/prerm-pkg"; echo "default_prerm"; ) > /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/pre-deinstall;
if [ -n "" ]; then echo  > /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.rusers; fi;
if [ -n "" ]; then echo  > /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.alternatives; fi;
(cd /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem && find . -type f,l -printf "/%P\n" > /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.list)
# Move conffiles to IDIR and build conffiles_static with csums
if [ -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/conffiles ]; then mv -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/apk-aarch64_cortex-a53/tom_modem/conffiles /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.conffiles; for file in $(cat /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.conffiles); do [ -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/$file ] || continue; csum=$(/home/qwer/immortalwrt-ipq/staging_dir/host/bin/mkhash sha256 /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/$file); echo $file $csum >> /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.conffiles_static; done; fi
# Some package (base-files) manually append stuff to conffiles
# Append stuff from it and delete the CONTROL directory since everything else should be migrated
if [ -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/CONTROL/conffiles ]; then echo $(IDIR_tom_modem)/CONTROL/conffiles >> /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.conffiles; for file in $(cat /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/CONTROL/conffiles); do [ -f /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/$file ] || continue; csum=$(/home/qwer/immortalwrt-ipq/staging_dir/host/bin/mkhash sha256 /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/$file); echo $file $csum >> /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/lib/apk/packages/tom_modem.conffiles_static; done; rm -rf /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/CONTROL/conffiles; fi
if [ -z "$(ls -A /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tom_modem-r0.9.2/ipkg-aarch64_cortex-a53/tom_modem/CONTROL 2>/dev/null)" ]; then rm -rf /home/qwer/immortalwrt-ipq/build_dir/target-aarch64_cortex-a53_musl/li [... cut ...]
time: package/feeds/modem/tom_modem/compile#0.16#0.01#0.16
    ERROR: package/feeds/modem/tom_modem failed to build.
make[2]: *** [package/Makefile:184: package/feeds/modem/tom_modem/compile] Error 1
make[2]: Leaving directory '/home/qwer/immortalwrt-ipq'
make[1]: *** [package/Makefile:178: /home/qwer/immortalwrt-ipq/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qwer/immortalwrt-ipq'
make: *** [/home/qwer/immortalwrt-ipq/include/toplevel.mk:248: world] Error 2

FUjr, 2024-12-10:
再试试呢？

qwer155168, 2024-12-10:
> 再试试呢？

好了，已经可以了，谢谢Tom哥

closed 2024-12-10 by qwer155168