FUjr/QModem #10: quectel-cm拨号主机名变bogon

RabbitYilia, 2024-12-18:
目前已知quectel-cm拨号主机名变bogon，拨号后重启br-lan可解决，原因还没搞明白，不知道怎么fix，作者有什么头绪嘛？

FUjr, 2024-12-19:
你好，我未发现相关代码，请问是什么条件下复现的呢？

RabbitYilia, 2024-12-19:
luci-app-adguardhome开启dnsmasq转发到adguardhome,如果是使用luci-proto-mbim拨号就没事，但是用quectel-CM-M拨号就会让内网主机名全变成bogon,但是自动重拨或者是重启br-lan就会恢复，手动点击重拨按钮就会触发。很奇怪。
[image]

RabbitYilia, 2024-12-19:
[image]
重启br-lan、创建出来的X_1接口就好了

RabbitYilia, 2024-12-19:
在openwrt-trunk分支，immoralwrt上均有复现

RabbitYilia, 2024-12-19:
> 你是哪个分支的哪个版本的openwrt呢？ 使用未经修改的quectel-CM会出现类似问题吗？
> […](#)

好吧，试了试未修改的quectel-CM，也会出现类似问题，我觉得和quectel-CM 修改DNS有关。

RabbitYilia, 2024-12-19:
那我先关闭吧，我看看quectel-CM的实现

closed 2024-12-19 by RabbitYilia

RabbitYilia, 2024-12-19:
破案了，看来还是要修改quectel-CM的实现。
nameserver 2408:8888:0:0:0:0:0:8 # IPV6 wwan0
nameserver 2408:8899:0:0:0:0:0:8 # IPV6 wwan0
nameserver 202.102.224.68 # IPV4 wwan0
nameserver 202.102.227.68 # IPV4 wwan0
search lan
nameserver 127.0.0.1
nameserver ::1

quectel-CM机械地添加nameserver到/etc/resolv.conf，但实际上新版的openwrt resolv.conf指向的是DNSMASQ，修改后的resolv.conf会把内网主机名解析发到运营商那里去，我写个contab 正则一下好了，quectel-CM我一时半会儿还真不知道怎么改。

FUjr, 2024-12-19:
已修复，请更新代码

@FUjr<https://github.com/FUjr>

commit 0545d9e4 (2024-12-19): add -D option to block modification of resolve.conf to fix #10