FUjr/QModem #11: Immortalwrt-24.10源码 为了配置IPV6在拨号配置列表，高级里勾选扩展前缀好像没用

czyjian, 2025-01-22:
Immortalwrt-24.10源码 为了配置IPV6在拨号配置列表，高级里勾选扩展前缀好像没用

不知道是24.10的兼容性问题还是咋？
虽然看readme是说21以上，请问是最推荐base哪个版本去用这个插件？

FUjr, 2025-01-22:
用的ecm/rndis/ncm吗？ipv6只有qmi/mbim可用，前者在模组内路由没有添加。

czyjian, 2025-01-22:
> 用的ecm/rndis/ncm吗？ipv6只有qmi/mbim可用，前者在模组内路由没有添加。
> […](#)

用的qmi/mbim 走PCIE的FM150AE

获取ipv6是没问题啦，地址有获取到。但是没有IPV6-PD，LAN只能用中继模式，PC终端也能获取到IPV6地址，但是无论如何IPV6都不通。蛮奇怪的。

然后发现Siriling 的那个插件可以获取IPV6-PD，并且LAN用客户端模式的话很完美~都是基于24.10编的。

然后对比了下两个IPV6 WAN的设置，也就这个扩展前缀不一样。

FUjr, 2025-01-22:
勾选扩展前缀后需要点一下重拨，更新拨号配置均需要重拨的

czyjian, 2025-01-22:
> 勾选扩展前缀后需要点一下重拨，更新拨号配置均需要重拨的
> […](#)
> 
> From: czyjian ***@***.***>
> Sent: Wednesday, January 22, 2025 11:41:28 AM
> To: FUjr/modem_feeds ***@***.***>
> Cc: Fu Jianrong ***@***.***>; Comment ***@***.***>
> 
> 
> 用的ecm/rndis/ncm吗？ipv6只有qmi/mbim可用，前者在模组内路由没有添加。
> …
> 
> 用的qmi/mbim 走PCIE的FM150AE
> 
> 获取ipv6是没问题啦，地址有获取到。但是没有IPV6-PD，LAN只能用中继模式，PC终端也能获取到IPV6地址，但是无论如何IPV6都不通。蛮奇怪的。
> 
> 然后发现Siriling 的那个插件可以获取IPV6-PD，并且LAN用客户端模式的话很完美~都是基于24.10编的。
> 
> 然后对比了下两个IPV6 WAN的设置，也就这个扩展前缀不一样。

勾选了之后有重播。然后去看也发现接口被删除后重新添加并播上了号。但是WAN IPV6的setting发现也没什么变化，如果手动直接去编辑接口的话，保存了之后就获取不到IPV6地址了。

czyjian, 2025-01-22:
> [image]![image](https://private-user-images.githubusercontent.com/53910283/405466450-4c8e9df9-408f-4b83-bdf7-48e8006a9bcf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc1MjU5MTYsIm5iZiI6MTczNzUyNTYxNiwicGF0aCI6Ii81MzkxMDI4My80MDU0NjY0NTAtNGM4ZTlkZjktNDA4Zi00YjgzLWJkZjctNDhlODAwNmE5YmNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTIyVDA2MDAxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI0MDU3NGU1NmRmOWJlMjIzMmUxOGIzYjk0OTQ2N2YxMWUzNGY5MDY1MjU4ZDUwZTRlMzUxOTMwNGQwODlkMTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.hA1fIQuIx7Oh-r-PQLPWGmQUYhKa4iWQysPZu9iE7fQ) 理论上这样设置就可以了，不过24的我还没测试过

设置好了之后，不小心重启了一下。。。发现确实出来了哈哈哈哈哈、、看来是要重启才行

不过我编译的时候发现单独拉你的插件源码，会少kmod-pcie_mhi和quectel-CM-5G
WARNING: Makefile 'package/feeds/modem/luci-app-qmodem/Makefile' has a dependency on 'kmod-pcie_mhi', which does not exist
WARNING: Makefile 'package/feeds/modem/luci-app-qmodem/Makefile' has a dependency on 'quectel-CM-5G', which does not exist

我拉的Siriling 的，看要不要补一下~

FUjr, 2025-01-22:
quectel-cm在不同feeds里有不同的命名

FUjr, 2025-01-22:
[ ] Using Tom customized Quectel CM                                       
[ ] Using QWRT quectel-CM-5G                                             
[ ] Using Normal quectel-cm   
我提供的修改过的quectel-cm是第一个，
qwrt feeds里的是第二个
主线（imm和openwrt）是第三个
三个只需要勾选一个就行了

FUjr, 2025-01-22:
至于pcie_mhi，由于版本较多，在mtk和高通的仓库可能有不同的补丁，我就没加上了

closed 2025-01-22 by czyjian