FUjr/QModem #19: 编译报错

Canary233, 2025-02-27:
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency luci-app-qmodem_with_lua_index_page for luci-app-qmodem
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-qmodem found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-qmodem.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-qmodem-zh-cn:
 *      luci-app-qmodem_with_lua_index_page
 * opkg_install_cmd: Cannot install package luci-i18n-qmodem-zh-cn.
```
config中有`CONFIG_PACKAGE_luci-app-qmodem_with_lua_index_page=y`
用的immortalwrt,openwrt-24.10分支.干净编译同样的报错

FUjr, 2025-02-28:
[image]
使用imm的24.10没能复现 理论上openwrt是一样的，如果问题依旧 可以尝试修改application/luci/luci-app-qmodem/Makefile 里，将Depends中的luci-app-qmodem_with_lua_index_page 移除

Canary233, 2025-02-28:
> [image] 使用imm的24.10没能复现 理论上openwrt是一样的，如果问题依旧 可以尝试修改application/luci/luci-app-qmodem/Makefile 里，将Depends中的luci-app-qmodem_with_lua_index_page 移除

把depends删了可以了

closed 2025-02-28 by Canary233