FUjr/QModem #21: 拉取最新代码后，使用modem_feeds的驱动，lede及immoralwrt固件编译时均报错，使用siriling的驱动暂时没有问题

yizhanghong, 2025-03-10:
错误代码如下：
make[5]: *** [scripts/Makefile.build:289: /home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_m-1.0/qmi_wwan_m.o] Error 1
make[4]: *** [Makefile:1907: /home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_m-1.0] Error 2
make[4]: Leaving directory '/home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173'
make[3]: *** [Makefile:50: /home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_m-1.0/.built] Error 2
make[3]: Leaving directory '/home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/feeds/5gsupport/meig_QMI_WWAN'
time: package/feeds/5gsupport/meig_QMI_WWAN/compile#0.52#0.32#0.82
    ERROR: package/feeds/5gsupport/meig_QMI_WWAN failed to build.
make[2]: *** [package/Makefile:129: package/feeds/5gsupport/meig_QMI_WWAN/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt'
make[1]: *** [package/Makefile:123: /home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt'
make: *** [/home/runner/work/AutoBuild-immortalwrt-23.05/AutoBuild-immortalwrt-23.05/openwrt/include/toplevel.mk:233: world] Error 2

FUjr, 2025-03-10:
meig的驱动未经测试，可以在luci/luci-app-qmodem/Makefile中移除依赖，然后在kernel module处取消meig驱动的勾选

sfwtw, 2025-03-15:
请提供更多信息，在lede，237和h大的imm-798x分支均未报错，如果是imm主线报错在[a4619fa](https://github.com/FUjr/modem_feeds/commit/a4619fa1ec9b2b2025798fb3c051dc89faed8781)已修复

closed 2025-03-17 by sfwtw