FUjr/QModem #26: 是否支持同时两个PCIE的5G模块

oldcat618, 2025-03-26:
机器插了两FM150-AE ，只能正常识别一个，都是PCIE模式

FUjr, 2025-03-26:
另一个at端口是mhi_DUN1吗？

FUjr, 2025-03-26:
设计时就已经加入了多个pcie模组的支持，但是未经测试。
刚刚调查了一下，有人反馈测试过可以正常加载多个pcie模组

oldcat618, 2025-03-27:
我机器两 pcie的FM150-AE只认一个，两个4G模块可以识别

FUjr, 2025-03-27:
lspci呢？以及/dev/下有哪些设备

oldcat618, 2025-03-27:
只有一个 mhi_DUN 另一个不工作

> lspci呢？以及/dev/下有哪些设备

oldcat618, 2025-03-30:
root@OpenWrt:~# lspci -v
00:00.0 PCI bridge: Qualcomm Technologies, Inc Device 1002 (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 22
        Memory at 20220000 (32-bit, non-prefetchable) [size=4K]
        Bus: primary=00, secondary=01, subordinate=ff, sec-latency=0
        I/O behind bridge: [disabled] [16-bit]
        Memory behind bridge: 20300000-203fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: [disabled] [64-bit]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/32 Maskable+ 64bit+
        Capabilities: [70] Express Root Port (Slot+), IntMsgNum 0
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Secondary PCI Express
        Capabilities: [158] Transaction Processing Hints
        Capabilities: [1ec] L1 PM Substates
        Kernel driver in use: pcieport
lspci: Unable to load libkmod resources: error -2

01:00.0 Unassigned class [ff00]: Qualcomm Technologies, Inc SDX55 [Snapdragon X55 5G]
        Subsystem: Device 1cf8:5502
        Flags: bus master, fast devsel, latency 0, IRQ 65
        Memory at 20300000 (64-bit, non-prefetchable) [size=4K]
        Memory at 20301000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=4/32 Maskable+ 64bit+
        Capabilities: [70] Express Endpoint, IntMsgNum 0
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Secondary PCI Express
        Capabilities: [168] Physical Layer 16.0 GT/s <?>
        Capabilities: [18c] Lane Margining at the Receiver
        Capabilities: [19c] Transaction Processing Hints
        Capabilities: [228] Latency Tolerance Reporting
        Capabilities: [230] L1 PM Substates
        Capabilities: [240] Data Link Feature <?>
        Kernel driver in use: mhi_q

FUjr, 2025-03-30:
只看到一个模组 换一个发行版试试看呢

closed 2025-04-06 by FUjr