FUjr/QModem #27: FM350-GL拨号，第二天IPV6失效的问题

aliass532, 2025-03-26:
大佬，FM350-GL使用RAX3000M拨号，刚开始IPV6正常，但第二天IPV6出现失效的现象，请辛苦看下原因，谢谢！

[image]
[image]
[image]
[拨号日志.log](https://github.com/user-attachments/files/19466295/default.log)
[内核日志.log](https://github.com/user-attachments/files/19466296/default.log)
[系统日志.log](https://github.com/user-attachments/files/19466297/default.log)

FUjr, 2025-03-26:
ipv4正常吗，以及在网络页面手动重启一下ipv6的接口有用吗？

aliass532, 2025-03-26:
ipv4正常，重启ipv6接口后正常

closed 2025-03-26 by FUjr

aliass532, 2025-03-30:
@FUjr 大佬，不好意思，出差2天刚反馈，在前天编译后，长时间后，IPV6还是会出现失效的现象，请辛苦再看下原因，谢谢！
[image]

FUjr, 2025-03-30:
有可能是，这段代码是社区贡献的 我没有详细测试，实际上变量 $interface6_name 即是ipv6接口名，可以修改后测试一下～

commit b52ceeea (2025-03-26): fix: fix fm350 ipv6 monitor. may resolve #27