FUjr/QModem #46: [BUG]

fdyf520, 2025-05-20:
以下是问题报告模板，包含中文和英文翻译：

---

**描述问题**  
提供模组型号、路由器平台、路由器系统发行版等信息  
rm500q-gl、zbt-z8102ax、openmptcprouter

mt7981方案

**复现方法**  
单独使用5G模块扩展板，接usb测试没有问题，模块和sim卡均能读取，lede编译的版本也不存在问题
现在zbt-z8102ax是nvme接口的，但是走的是usb通道。目前测试单个模块短时间能进行拨号，并且能够拨号成功。但是两个模块同时使用的时候，会出现sim卡未插入的情况，手动软重启模块，能够识别，但是一分钟左右系统就不能识别了。出现sim miss的问题，界面均无法改动和操作

---

**期望行为**  
描述正常情况下应该是什么行为  
修复bug，或者提供解决思路

---

**屏幕截图**  
**Screenshots**  

[image]

---

**日志信息**  
请在终端执行以下命令并将结果粘贴到此处  
**Log Information**  
Please execute the following commands in the terminal and paste the results below:

```bash
uci show qmodem
# 显示模组配置
# Show module configuration

uci show network
# 显示网络配置
# Show network configuration

logread
# 查看系统日志
# View system logs

dmesg
# 查看内核日志
# View kernel logs
```

如果是USB模组相关的问题，执行以下命令：  
**If the issue is related to USB modules, execute the following command:**

```bash
lsusb
# 列出USB设备
# List USB devices
```

如果是PCIe模组相关的问题，执行以下命令：  
**If the issue is related to PCIe modules, execute the following command:**

```bash
lspci
# 列出PCIe设备
# List PCIe devices
```

如果模组扫描存在问题，执行以下命令：  
**If there is an issue with module scanning, execute the following command:**

```bash
/usr/share/qmodem/modem_scan.sh scan
# 扫描模组
# Scan modules
```

---

fdyf520, 2025-05-21:
This model is a multi-card router that has been adapted for OpenWrt and compiled with Qmodem on top of it. I'm not sure if the issue lies with the USB bridge module or the SIM card configuration. When no SIM card is inserted, the function works perfectly fine. However, as soon as a SIM card is inserted, it shows "SIM miss". I haven't encountered this problem with an external expansion board, so I'm thinking it might be a hardware issue or a software compatibility problem.

FUjr, 2025-05-21:
请把模板中需要填写的部分完善

fdyf520, 2025-05-21:
以下是问题报告模板，包含中文和英文翻译：

---

**描述问题**  
提供模组型号、路由器平台、路由器系统发行版等信息  
**Describe the Issue**  
Provide module model, router platform, router system distribution, etc.

---

**复现方法**  
**Steps to Reproduce**  
1. Go to '...'  
2. Click on '...'  
3. Scroll down to '...'  
4. See the error

---

**期望行为**  
描述正常情况下应该是什么行为  
**Expected Behavior**  
Describe what the normal behavior should be.

---

**屏幕截图**  
**Screenshots**  

---

**日志信息**  
请在终端执行以下命令并将结果粘贴到此处  
**Log Information**  
Please execute the following commands in the terminal and paste the results below:

```bash
uci show qmodem
# 显示模组配置
# Show module configuration

root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: DNS service limited to local subnets
 p-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
-ash: Tue: not found
d: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Tue May 20 15:54:53 2025 daemon.notice procd: /etc/rc.d/S19firewall: root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
 j-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
:-ash: Tue: not found
root@FDYF-MPTCP:~# Tue May 20 15:54:51 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
[... 2064 lines cut ...]

```bash
/usr/share/qmodem/modem_scan.sh scan
# 扫描模组
# Scan modules
```
ash: 1: unknown operand
ash: 1: unknown operand
ash: 1: unknown operand
ash: 1: unknown operand
[ERR]Failed to run operation 1
[ERR]Error restoring old tty attributes
[ERR]Failed to run operation 1
[ERR]Error restoring old tty attributes
[ERR]Failed to run operation 1
[ERR]Error restoring old tty attributes
ash: 1: unknown operand
ash: 1: unknown operand
ash: 1: unknown operand
ash: 1: unknown operand
[ERR]Failed to run operation 1
[ERR]Error restoring old tty attributes
[ERR]Failed to run operation 1
[ERR]Error restoring old tty attributes
---

fdyf520, 2025-05-21:
[image]

sfwtw, 2025-05-31:
切换sms_tools应该会改善部分情况 maybe tom_modem尚未完善 [image]

fdyf520, 2025-05-31:
这个文件是那个呢

sfwtw, 2025-05-31:
> 这个文件是那个呢

/luci/luci-app-qmodem/root/usr/share/qmodem/modem_util.sh

fdyf520, 2025-06-03:
sms_tools 这个指的是luci-app-sms_tools吗，但是我安装了并且也替换了.sh。但是扫描不到模块了

closed 2025-07-05 by sfwtw