FUjr/QModem #47: [BUG]和docker网络冲突

missorforget, 2025-05-22:
平台：x86
模组：rm500q
docker安装、启动小雅时

closed 2025-05-22 by missorforget

missorforget, 2025-05-22:
code:500 failed to set up container networking: driver failed programming external connectivity on endpoint xiaoya(...); Unable to enable DNAT rule:(iptables failed: iptables --wait -t nat -A DOCKER -p tcp -d 0/0 --dport 5678 -j DANT --to-destination 172.17.0.2:80 ! -i docker0: iptables: No chain/target/match by that name.(exit status 1)))

FUjr, 2025-05-22:
1.  什么分支的什么版本？fw是fw3还是fw4？
2.  如果不编译安装此插件，是否会出现这个报错？
3.  执行/etc/init.d/docker restart 重启docker是否还会出现

missorforget, 2025-05-22:
1. lean的源码，fw3
2. 不安装没有这个报错
3. 会出现

FUjr, 2025-05-22:
QModem会重载防火墙，/etc/init.d/firewall restart,重载后会清理所有规则，包括docker创建的chain。理论上openwrt的docker软件包会在重载后重新创建相应chain，可以看看不使用QModem时重载firewall 是否会引发相同问题，会的话则建议在docker端处理

missorforget, 2025-05-22:
好的，谢谢

FUjr, 2025-05-22:
> 验证了一下，不使用qmodem运行/etc/init.d/firewall restart可以复现问题，证实了firewall重载后docker没有重新创建相应chain

我不怎么在Op使用docker，但是docker应该会自动添加用户脚本，fw重载后仍能恢复防火墙规则，建议换一个分支测试