FUjr/QModem #75: [BUG]

grayliy, 2025-07-23:
**描述问题**  
提供模组型号、路由器平台、路由器系统发行版等信息  
**Describe the Issue**  
Provide module model, router platform, router system distribution, etc.

FM350-GL

HINLINK H29K Board

LuCI openwrt-23.05 branch git-25.110.68714-844a8ee 

6.6.88
--

**复现方法**  
**Steps to Reproduce**  

多次重启后，拨号总览会出现两个拨号状态，且主页总览，也会有两个信号条

[image]

[image]

---

**期望行为**  
描述正常情况下应该是什么行为  
**Expected Behavior**  
Describe what the normal behavior should be.

拨号总览 只有一个状态，主页总览，也只有一个模组信息
---

**屏幕截图**  
**Screenshots**  

---

**日志信息**  
请在终端执行以下命令并将结果粘贴到对应的折叠代码块中  
**Log Information**  
Please execute the following commands in the terminal and paste the results into the corresponding collapsible code blocks below:

<details>
<summary>模组配置 (Module Configuration)</summary>

```bash
uci show qmodem
# 显示模组配置
# Show module configuration
```

粘贴结果到此处  
Paste the output here
```
root@H29K:~# uci show qmodem
qmodem.main=main
qmodem.main.enable_dial='1'
qmodem.main.try_preset_usb='1'
qmodem.main.try_preset_pcie='1'
qmodem.main.enable_usb_scan='1'
qmodem.main.modem_count='2'
qmodem.3_1=modem-device
qmodem.3_1.path='/sys/bus/usb/devices/3-1/'
qmodem.3_1.data_interface='usb'
qmodem.3_1.enable_dial='1'
qmodem.3_1.soft_reboot='1'
qmodem.3_1.extend_prefix='1'
qmodem.3_1.pdp_type='ipv4v6'
qmodem.3_1.state='enabled'
qmodem.3_1.metric='11'
qmodem.3_1.name='fm350-gl'
qmodem.3_1.network='eth1'
qmodem.3_1.manufacturer='fibocom'
qmodem.3_1.platform='mediatek'
qmodem.3_1.define_connect='3'
qmodem.3_1.wcdma_band='null'
qmodem.3_1.lte_band='null'
qmodem.3_1.nsa_band='null'
qmodem.3_1.sa_band='null'
qmodem.3_1.modes='rndis'
qmodem.3_1.valid_at_ports='/dev/ttyUSB1' '/dev/ttyUSB3'
qmodem.3_1.at_port='/dev/ttyUSB3'
qmodem.3_1.ports='/dev/ttyUSB0' '/dev/ttyUSB1' '/dev/ttyUSB2' '/dev/ttyUSB3' '/dev/ttyUSB4' '/dev/ttyUSB5' '/dev/ttyUSB6'
qmodem.4_1=modem-device
qmodem.4_1.path='/sys/bus/usb/devices/4-1/'
qmodem.4_1.data_interface='usb'
qmodem.4_1.enable_dial='1'
qmodem.4_1.soft_reboot='1'
qmodem.4_1.extend_prefix='1'
qmodem.4_1.pdp_type='ipv4v6'
qmodem.4_1.state='enabled'
qmodem.4_1.metric='12'
qmodem.4_1.name='fm350-gl'
qmodem.4_1.network='eth1'
qmodem.4_1.manufacturer='fibocom'
qmodem.4_1.platform='mediatek'
qmodem.4_1.define_connect='3'
qmodem.4_1.wcdma_band='null'
qmodem.4_1.lte_band='null'
qmodem.4_1.nsa_band='null'
qmodem.4_1.sa_band='null'
qmodem.4_1.modes='rndis'
qmodem.4_1.valid_at_ports='/dev/ttyUSB1' '/dev/ttyUSB3'
qmodem.4_1.at_port='/dev/ttyUSB3'
qmodem.4_1.ports='/dev/ttyUSB0' '/dev/ttyUSB1' '/dev/ttyUSB2' '/dev/ttyUSB3' '/dev/ttyUSB4' '/dev/ttyUSB5' '/dev/ttyUSB6'

```
</details>

<details>
<summary>网络配置 (Network Configuration)</summary>

```bash
uci show network
# 显示网络配置
# Show network configuration
```
root@H29K:~# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.packet_steering='1'
network.globals.ula_prefix='fd1f:44a3:7260::/48'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='eth0'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.1'
network.lan.netmask='255.255.255.0'
network.lan.ipv6='1'
network.lan.ip6assign='64'
network.docker=interface
network.docker.device='docker0'
network.docker.proto='none'
network.docker.auto='0'
network.@device[1]=device
network.@device[1].type='bridge'
network.@device[1].name='docker0'
network.4_1=interface
network.4_1.modem_config='4_1'
network.4_1.proto='static'
network.4_1.defaultroute='1'
network.4_1.metric='12'
network.4_1.ifname='eth1'
network.4_1.device='eth1'
network.4_1.ipaddr='10.100.76.163'
network.4_1.netmask='255.255.255.0'
network.4_1.gateway='10.100.76.1'
network.4_1.peerdns='0'
network.4_1.dns='223.5.5.5' '119.29.29.29'
network.4_1v6=interface
network.4_1v6.modem_config='4_1'
network.4_1v6.proto='dhcpv6'
network.4_1v6.ifname='@4_1'
network.4_1v6.device='@4_1'
network.4_1v6.metric='12'
network.4_1v6.extendprefix='1'

粘贴结果到此处  
Paste the output here
```
```
</details>

<details>
<summary>系统日志 (System Logs)</summary>

```bash
logread
# 查看系统日志
# View system logs
```

粘贴结果到此处  
Paste the output here
```
```
</details>

<details>
<summary>内核日志 (Kernel Logs)</summary>

```bash
dmesg
# 查看内核日志
# View kernel logs
```

粘贴结果到此处  
Paste the output here
```
```
</details>

如果是USB模组相关的问题，执行以下命令并填写结果：  
**If the issue is related to USB modules, execute the following command and paste the output:**

<details>
<summary>USB设备信息 (USB Devices)</summary>

```bash
lsusb
# 列出USB设备
# List USB devices
```

[image]

粘贴结果到此处  
Paste the output here
```
```
</details>

如果是PCIe模组相关的问题，执行以下命令并填写结果：  
**If the issue is related to PCIe modules, execute the following command and paste the output:**

<details>
<summary>PCIe设备信息 (PCIe Devices)</summary>

```bash
lspci
# 列出PCIe设备
# List PCIe devices
```

粘贴结果到此处  
Paste the output here
```
```
</details>

如果模组扫描存在问题，执行以下命令并填写结果：  
**If there is an issue with module scanning, execute the following command and paste the output:**

<details>
<summary>模组扫描 (Module Scanning)</summary>

```bash
/usr/share/qmodem/modem_scan.sh scan
# 扫描模组
# Scan modules
```

粘贴结果到此处  
Paste the output here
```
```
</details>

FUjr, 2025-07-23:
看看内核日志 和系统日志 看起来像是udb2.0 和usb3.0分别扫了一次

grayliy, 2025-07-27:
> 看看内核日志 和系统日志 看起来像是udb2.0 和usb3.0分别扫了一次

如果这个编号是随机变动的话，就不方便保存了，最好是每次开机重新扫描，然后使用最新的编号。

handsouls, 2025-08-12:
这个BUG在切换拨号模式的时候也会复现

FUjr, 2025-08-16:
可以看看这个分支是否修复https://github.com/FUjr/QModem/tree/fix/fuck_h29k