FUjr/QModem #92: [BUG]使用物联网卡时设备在每次重启后无法上网

1312564541, 2025-08-14:
**描述问题**  
硬件：airpi ap3000m
固件版本：QWRT R25.8.8 / LuCI Master (git-25.224.25455-af12e8d)  
5G 模组：RM502QAEAAR13A04M4G
QModem版本：2.8.5
号卡：联通物联网卡
问题：每次重启设备后无法上网，必须手动修改一次APN且不能跟原来一样，再重拨才能上网，关键的日志如下面的“no need to set skip the rest”：
[08-13_17:40:03:533] requestGetProfile[pdp:1 index:1] 3gnet///0/IPV4V6
[08-13_17:40:03:533] requestSetProfile[pdp:1 index:1] 3gnet///0/IPV4V6
[08-13_17:40:03:533] no need to set skip the rest

---

**复现方法**  
假设当前可以正常上网，APN设置那里选择的是“自动”
1、重启设置，无法上网，日志如下：
[image]

2、修改APN为中国联通，保存应用，重拨，恢复上网
[image]

3、重启设备，无法上网
[image]

4、按步骤2操作，恢复上网
[image]

---

**期望行为**  
猜测是前后两个APN一致，跳了某个环节从而导致了问题的出现，这个环节可能是设置APN的，据网友反馈物联网卡必须配置APN才能上网，而正常号卡不需要，期待修正。
临时措施，在QModem设置里，编辑模组，在拨号前AT命令里依次添加AT+CGACT=0,1和AT+CGACT=1,1，后面重新重启之后都能正常上网了。

---

**屏幕截图**  
**Screenshots**  

---

**日志信息**  
请在终端执行以下命令并将结果粘贴到对应的折叠代码块中  
**Log Information**  
Please execute the following commands in the terminal and paste the results into the corresponding collapsible code blocks below:

<details>
<summary>模组配置 (Module Configuration)</summary>

```bash
uci show qmodem
# 显示模组配置
# Show module configuration
```

粘贴结果到此处  
Paste the output here
```
root@QWRT:~# uci show qmodem
qmodem.main=main
qmodem.main.enable_dial='1'
qmodem.main.try_preset_usb='1'
qmodem.main.try_preset_pcie='1'
qmodem.main.enable_pcie_scan='1'
qmodem.main.start_delay='0'
qmodem.main.modem_count='1'
qmodem.main.enable_usb_scan='1'
qmodem.2_1=modem-device
qmodem.2_1.path='/sys/bus/usb/devices/2-1/'
qmodem.2_1.data_interface='usb'
qmodem.2_1.enable_dial='1'
[... 452 lines cut ...]

```
</details>

如果是USB模组相关的问题，执行以下命令并填写结果：  
**If the issue is related to USB modules, execute the following command and paste the output:**

<details>
<summary>USB设备信息 (USB Devices)</summary>

```bash
lsusb
# 列出USB设备
# List USB devices
```

粘贴结果到此处  
Paste the output here
```
root@QWRT:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.101 xhci-hcd xHCI Host Controller
Bus 002 Device 002: ID 2c7c:0800 Quectel RM502Q-AE
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.101 xhci-hcd xHCI Host Controller
```
</details>

FUjr, 2025-08-14:
https://github.com/FUjr/QModem/actions/runs/16958903498 
登录github后在这里下载 ，把quectel-CM-5G-M 和qmodem、luci-app-qmodem 的ipk安装一下，或者解压对应ipk 并替换文件，

1312564541, 2025-08-15:
感谢大佬的处理，刚下载文件安装了，单个ipk文件挨个安装，再全部安装一次，ubus-at-daemon无法安装（缺少3个依赖（libubus20250702、libubox20250723、libblobmsg-json20250723），网上找过似乎没有这3个的ipk文件，只有apk文件，当前immortalwrt系统不支持apk），其他安装了，重启设备，仍没能解决问题，日志如下：
[08-15_09:47:20:704] requestBaseBandVersion RM502QAEAAR13A04M4G
[08-15_09:47:20:704] Modem model RM502QAEAAR13A04M4G does not require FCC authentication
[08-15_09:47:20:736] qmap_settings.rx_urb_size = 31744
[08-15_09:47:20:736] qmap_settings.ul_data_aggregation_max_datagrams  = 11
[08-15_09:47:20:736] qmap_settings.ul_data_aggregation_max_size       = 4096
[08-15_09:47:20:736] qmap_settings.dl_minimum_padding                 = 0
[08-15_09:47:20:896] Skipping FCC authentication - not required for this modem model
[08-15_09:47:20:928] requestGetSIMStatus SIMStatus: SIM_READY
[08-15_09:47:20:992] requestGetProfile[pdp:1 index:1] auto///0/IPV4V6
[08-15_09:47:20:992] requestSetProfile[pdp:1 index:1] auto///0/IPV4V6
[08-15_09:47:20:992] Force APN setting enabled, proceeding with profile update
[08-15_09:47:21:056] requestRegistrationState2 MCC: 460, MNC: 1, PS: Attached, DataCap: 5G_SA
[08-15_09:47:21:088] requestQueryDataCall IPv4ConnectionStatus: DISCONNECTED
[08-15_09:47:21:120] requestQueryDataCall IPv6ConnectionStatus: DISCONNECTED
[08-15_09:47:21:120] ip link set dev wwan0 down
[08-15_09:47:21:124] ip addr flush dev wwan0_1
[08-15_09:47:21:128] ip link set dev wwan0_1 down
[08-15_09:47:21:152] requestSetupDataCall QMUXResult = 0x1, QMUXError = 0xe
[08-15_09:47:21:152] call_end_reason is 1
[08-15_09:47:21:152] call_end_reason_type is 2
[08-15_09:47:21:152] call_end_reason_verbose is 241
[08-15_09:47:21:184] requestSetupDataCall QMUXResult = 0x1, QMUXError = 0xe
[08-15_09:47:21:184] call_end_reason is 1
[08-15_09:47:21:184] call_end_reason_type is 2

[image]

FUjr, 2025-08-16:
[github action](https://github.com/FUjr/QModem/actions/runs/17003128671)
加了清理apn的逻辑，只替换quectel-cm的包试验一下
观察是否出现clear APN settings相关日志，如果出现了这条日志还不行的话可能就比较复杂了

1312564541, 2025-08-19:
感谢大佬，现在断电启动，以及重启，都可以正常上网了。

操作过程：
1、first boot 恢复出厂设置；
2、把https://github.com/FUjr/QModem/actions/runs/16958903498里的安装包再加上https://github.com/FUjr/QModem/actions/runs/17009256751里quectel-cm的2.8.6的包，上传到路由器；
3、全部安装，有几个提示安装失败，图里有记录，似乎不影响；
4、过一会，刷新页面，勾上强制重设vpn，保存应用；
5、断电，启动，能上网；
6、web界面重启，能上网。

[image]